TelluIoT/ThingML #222: Timer protocol does not work for posix

antons1, 2018-06-06:
It seems like the Timer protocol does not work for posix.

Given this model:

```
thing fragment TimerMsgs {
	message second() @xms_tic "1000"
}

thing fragment TimerClient includes TimerMsgs {
	required port TimerPort {
		receives second
	}
}

thing TimerThing includes TimerClient {
	statechart init Timing {
		state Timing {
			on entry `fprintf(stderr, "%s", "Starting...\n");`
			internal event TimerPort?second
			action `fprintf(stderr, "%s", "This was a second.\n");`
		}
	}
}

protocol Timer;

configuration TimerCfg {
	instance tt: TimerThing
	connector tt.TimerPort over Timer
}
```

I would expect the compiled program to print `This was a second.` every second, but it only prints `Starting...`, and then does nothing.

brice-morin, 2018-06-06:
as a temporary work around, you can try the [good old timers](https://github.com/HEADS-project/training/blob/master/1.ThingML_Basics/4.Timers/_c/CTimer.thingml)

brice-morin, 2018-10-24:
timers are now to be used through the STL `import "timers.thingml" from stl`

closed 2018-10-24 by brice-morin