TelluIoT/ThingML #223: Better array support

jakhog, 2018-06-12:
For arrays to be more useful, I would like to have:

1. For-each loops (like we have discussed in #160). I feel like I have made the mistake of forgetting to increment the index in while loops enough times now. Since we also know the cardinality of the arrays at compile-time, this should be an easy thing to do.

2. Array initializers, e.g. `var array : Integer[3] = { 1, 3, 5, }`. Yes, it is mostly sugar and just to reduce copy-pasting of `set` statements. But it also helps to make sure you actually set all the values (and not to many) of the array.

3. Array slicing. So you don't have to manually copy values if you want to pass part of an array to e.g. a function. E.g. `var a : Int[100]`, then you could do, `var b : Int[10] = a[0:10]`.

closed 2018-06-26 by brice-morin

jakhog, 2018-07-05:
No slicing? 😢

brice-morin, 2018-07-05:
Seems we already are too fancy....