TelluIoT/ThingML #235: JavaScript TTY plugin requires the process to run inside an actual TTY

piernov, 2018-07-09:
In f64b280aa841c3cc025d237c588e154f4ab6022c the TTY plugin for JavaScript was made to require an actual TTY.
This prevents use outside shell, for example calling from a NodeJS application using `child_process.fork()` and then redirecting input/ouput streams to exchange message with the ThingML process or between multiple ThingML processes.

Any reason a TTY would be needed? If not, could this be reverted?

Thanks,

brice-morin, 2018-07-09:
A key use case for the TTY plugin is for debugging purpose, together with the JSON serialization plugin, where the user would type some JSON payloads in the terminal. I would not use the TTY plugins in a more advanced way.

piernov, 2018-07-09:
What would be the preferred way for IPC, or for integrating with NodeJS?

brice-morin, 2018-07-09:
Well, if two things implemented in ThingML are running locally on the same node (in the same ThingML configuration), there is no need for using external connectors and protocols (just normal connectors). Otherwise, the UDP or Websocket plugins could be used. But I must confess we have not used them for a while, so it might be some bug fixing to be done there.

closed 2018-12-11 by brice-morin