TelluIoT/ThingML #236: RaspberryPI - To Blink LED

brianrachel94, 2018-07-11:
Hi. I  want to blink my LED connected to the RaspberryPI. I have given my Thingml code below. I am using the 4th pin in RaspberryPi to connect to LED and how to mention that it my ThingML code? I don't know how to blink LED using the ThingML code in RaspberryPI. Can u pls help me on this?

Note:
Couldn't I install Eclipse in RaspberryPI ? is there a way to install eclipse in it? I install eclipse in my laptop and compile the ThingML code in Nodejs (with docker image) and copy the generated folder to RaspberryPI and run the Docker container. Is there any other easy way to do it???????????

/***********To Blink LED *************************************/
configuration myled
@docker "true"
{
instance helloLed : LedBlink
}

thing fragment TimerMsgs {
	// Start the Timer
	message timer_start();
	// Notification that the timer has expired
	message timer_timeout();
	//toggle the led for a particular frequency
	message led_toggle();
}

thing LedBlink includes TimerMsgs

{
	//readonly property LED : UInt8 = 8 // pin where the LED is plugged
	optional required port Timer
	{
		sends timer_start
        receives timer_timeout
	}
	
	optional required port LedTimer
    {	
		sends led_toggle
	}

	
	statechart Led init Blink
	{
		
	  state Blink
	  {
	  	
	   on entry Timer!timer_start ()
	   transition toggle -> Blink
	      event Timer?timer_timeout
	   action LedTimer!led_toggle()
	  }
    }
}

/********************END******************/

brice-morin, 2018-07-11:
Use MRAA https://github.com/intel-iot-devkit/mraa

brianrachel94, 2018-07-12:
I saw about MRAA. But how can this solve my problem? Can you please explain?

ffleurey, 2018-07-12:
Hello, lets take a step back here. 

There is no magic with ThingML, if you do not know how to make the LED blink on your target platform (i.e. using NodeJS on a Raspberry Pi), you have no chance of making it work in ThingML. I am not working much with NodeJS but there seem to be a fair amount of examples and tutorial online (just google "nodejs led blink raspberry pi"). Once you have figured out what library and instruction you can use to control the GPIOs you will be able to put those in your ThingML program and it will work just fine.

Concerning putting Eclipse on the RPI, you could try but that is really not the way to go. Eclipse should be on your development machine (laptop), then you generate the code and deploy it on your PI. You can automate that with scripts if you think it is tedious. In production we do it with ansible.

Good luck with your project!

brianrachel94, 2018-07-12:
Thanks a lot.

brianrachel94, 2018-07-12:
Below is the code to Blink the LED(referred from W3 schools) and I was able to implement it in RaspberryPI.Here for example, here I have used "onoff module" in RaspberryPI to blink the LED. How to declare the sample in ThingML? 

Can u post at least one example fro RaspberryPI so that it would be very useful to use ThingML and RaspberryPI, please?
 
/**********to blink Led****************/
var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
var LED = new Gpio(4, 'out'); //use GPIO pin 4, and specify that it is output
var blinkInterval = setInterval(blinkLED, 250); //run the blinkLED function every 250ms

function blinkLED() { //function to start blinking
  if (LED.readSync() === 0) { //check the pin state, if the state is 0 (or off)
    LED.writeSync(1); //set pin state to 1 (turn LED on)
  } else {
    LED.writeSync(0); //set pin state to 0 (turn LED off)
  }
}

function endBlink() { //function to stop blinking
  clearInterval(blinkInterval); // Stop blink intervals
  LED.writeSync(0); // Turn LED off
  LED.unexport(); // Unexport GPIO to free resources
}

setTimeout(endBlink, 5000); //stop blinking after 5 seconds
********************/

/************To blink led*********************************/
import "_Datatypes.thingml"

thing LEDMsgs {
	message led_ON();
	message led_OFF();
}

thing LED includes LEDMsgs {

	property PIN: UInt8 = 4

	provided port ctrl {
		receives led_ON, led_OFF
	}
	
	function setDigitalOutput(pin: UInt8) do
		`pinMode(`&pin&`,OUTPUT);`
	end
	
	function digitalWrite(pin: UInt8, value : DigitalState) do
		`digitalWrite(`&pin&`, `&value&`);`
	end
	
	statechart LED init READY {
		on entry setDigitalOutput(PIN)
		state READY {
			internal event ctrl?led_ON action digitalWrite(PIN, DigitalState:HIGH)
			internal event ctrl?led_OFF action digitalWrite(PIN, DigitalState:LOW)
		}
	}
}
/***************************************************/

ffleurey, 2018-07-14:
Discussion continued in #237.

closed 2018-07-14 by ffleurey