TelluIoT/ThingML #237: ThingML for my project

brianrachel94, 2018-07-12:
I am learning ThingML for the first Time and I am still not sure how to code in it for my project.

 My project is to to get data from the sensors(sensor like temp, pressure etc connected to raspberryPI) from the client using socket/WebSocket as the mode of communication. Whenever the value of sensor changes that value should go to the client.  I want to use ThingML to receive data asynchronously. But I don't know how to code in ThingML. is there any tutorial to learning ThingML programming. 

Please help me with this.

ffleurey, 2018-07-13:
There is currently not any extensive up-to-date documentation / tutorials but there are a lot of bits and pieces. The README at https://github.com/TelluIoT/ThingML is the place to start. It has an up-to-date set of links to a tutorial and some samples. We are working on better documentation and support but it will take some time.

For the RPI, I pushed a couple of simple blink examples to get you started: https://github.com/ffleurey/ThingML-RPI-Blink

If you make anything which you think could help others get started we will gladly link to it or include it in some samples repository.

brianrachel94, 2018-07-16:
Thanks a lot. My LED blinks using ThingML. Now I will try to implement ThingML for all the sensors which I am using in my Project.

closed 2018-07-16 by brianrachel94