TelluIoT/ThingML #240: Code generation for C compilers

imberium, 2018-07-24:
Hi, 
I am trying to generate some C or Arduino code from the following ThingML code, but I am getting the error below, I don't know what is missing, everything seems correct, it generates code for the other languages though like Java or ReactJS :
``thing Hello
{	
	statechart HelloBehavior init setup {

		 state setup 
		 @ignore "sink"
		 {
		 	on entry 
		 	do 
		 		print "Hello"
		 	end
		 	}
		 	
		}
}

configuration monitorCfg
{
	instance hello:Hello
}``

Please contact the ThingML development team (though GitHub's issue tracker) with 1) your input model, and 2) the following stack trace:
FATAL ERROR: Exeption calling ThingML Compiler: java.lang.NullPointerException
	org.thingml.compilers.c.CThingApiCompiler.generateCHeader(CThingApiCompiler.java:98)
	org.thingml.compilers.c.CThingApiCompiler.generatePublicAPI(CThingApiCompiler.java:83)
	org.thingml.compilers.c.posixmt.PosixMTCompiler.do_call_compiler(PosixMTCompiler.java:71)
	org.thingml.compilers.utils.OpaqueThingMLCompiler.compile(OpaqueThingMLCompiler.java:120)
	org.thingml.compilers.ThingMLCompiler.compile(ThingMLCompiler.java:310)
	org.thingml.eclipse.ui.commands.CompileThingFile.execute(CompileThingFile.java:255)
	org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:291)
	org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:93)
	sun.reflect.GeneratedMethodAccessor90.invoke(Unknown Source)
	sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	java.lang.reflect.Method.invoke(Method.java:498)
	org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:161)
	org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:152)
	org.eclipse.core.commands.Command.executeWithChecks(Command.java:494)
	org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:204)
	org.eclipse.ui.internal.handlers.LegacyHandlerService.executeCommand(LegacyHandlerService.java:393)
	org.eclipse.ui.menus.CommandContributionItem.handleWidgetSelection(CommandContributionItem.java:820)
	org.eclipse.ui.menus.CommandContributionItem.lambda$3(CommandContributionItem.java:797)
	org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	org.eclipse.swt.widgets.Display.sendEvent(Display.java:5686)
	org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1370)
	org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4940)
	org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4518)
	org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1170)
	org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1059)
	org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:667)
	org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:597)
	org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	java.lang.reflect.Method.invoke(Method.java:498)
	org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:656)
	org.eclipse.equinox.launcher.Main.basicRun(Main.java:592)
	org.eclipse.equinox.launcher.Main.run(Main.java:1498)
	org.eclipse.equinox.launcher.Main.main(Main.java:1471)

brice-morin, 2018-07-25:
I cannot reproduce this issue. Can you give more details about your setup (OS, Eclipse Version, ThingML plugins version, etc)

imberium, 2018-07-25:
Here is my configuration, I use Eclipse Photon, but I remember having the same problem with Oxygen
Maven  3.5.3
Java 1.8.0_151, vendor: Oracle Corporation
Default locale: en_US, platform encoding: UTF-8
OS name: "linux", version: "4.15.0-24-generic", arch: "amd64", family: "unix"

imberium, 2018-07-25:
However, everything works fine when I use the CLI version from the repo. Maybe the released version need to be updated

closed 2019-04-09 by brice-morin