TelluIoT/ThingML #241: ThingML as a maven dependency

imberium, 2018-07-25:
Hi,

There is an issue when using ThingML as a maven dependency, maven fails to import it, it seems that there is a problem in the repository. I get this error :

```Failed to collect dependencies at org.thingml:compilers.registry:jar:2.0.0-SNAPSHOT -> org.thingml:compilers.c:jar:2.0.0-SNAPSHOT -> org.thingml:compilers.framework:jar:2.0.0-SNAPSHOT -> org.thingml:thingml:jar:2.0.0-SNAPSHOT: Failed to read artifact descriptor for org.thingml:thingml:jar:2.0.0-SNAPSHOT: Could not find artifact org.thingml:language:pom:2.0.0-SNAPSHOT in thingml-snapshot```

brice-morin, 2018-07-25:
Hi,

We used to have a Maven repo, but unfortunately, it is not maintained/available anymore... @ffleurey when do you plan to have a public Maven repo available?

I am trying a solution based on [jitpack.io](https://jitpack.io/), but I cannot (yet) make it work... I'll keep you updated if I succeed. Maybe you can also try jitpack? Seems like a neat solution if we can get it to work...

brice-morin, 2018-07-25:
As an alternative to jitpack.io, I could make a [Maven repo into a mvn-repo branch ](https://stackoverflow.com/questions/14013644/hosting-a-maven-repository-on-github) on the ThingML repo... what do you think @ffleurey ???

imberium, 2018-07-25:
I ended up compiling the source code directly from the repo and adding the jar to my project, 
```            <dependency>
			<groupId>org.thingml</groupId>
			<artifactId>thingml-registry</artifactId>
			<version>1.0</version>
			<scope>system</scope>
			<systemPath>${project.basedir}/path/to/thingml-registry.jar</systemPath>
		</dependency>``` 

I suggest that you remove the maven option from the readme (as it is not working)

brice-morin, 2019-08-20:
OK, we should probably look into this: https://help.github.com/en/articles/about-github-package-registry#supported-clients-and-formats