TelluIoT/ThingML #243: New Compile buttons in Eclipse do not work with network plugins?

ffleurey, 2018-09-28:
When trying to compile breakout for posix (ThingML-PongTutorial/src/breakout/_posix/BreakoutPosix.thingml), it works when using the old HEADS menu but with the new one the timer plugin is not generated. Strange.

Olso the "Generare and run" option does not seem to run anything (tested with posix and java).

jakhog, 2018-09-29:
The "Generate and Run" only "Generate"s so far, the "Run" has not been implemented yet. I might very well have forgotten something about the network plugins and how to initialize them. I will have a pass at it soon, I would assume that all we have to do is look at the HEADS-menu, and the CLI code and do the same in the LaunchConfiguration...

brice-morin, 2018-09-29:
I haven't checked, but those menus should wrap official-network-plugins,
and not directly the registry.

commit ac248f04 (2018-10-01): Use registry to create instance instead of .clone() in launcher. Fixes #243.

The ThingMLCompiler.clone() currently doesn't propagate plugins, so the
previous ThingMLLauncher used in the new Eclipse UI buttons didn't
compile NetworkPlugins. Now it does.
closed 2018-10-01 by jakhog