TelluIoT/ThingML #247: The brower compiler uses the non standard setImmediate

fungiboletus, 2018-11-01:
`setImmediate` is a non-standard method only available in Internet Explorer, Edge, and NodeJS. Polyfils such as [YuzuJS/setImmediate](https://github.com/YuzuJS/setImmediate) do exist. A `setTimeout` could be used instead, but it may have a 4ms delay.

`requestIdleCallback` and `requestAnimationFrame` are also alternatives to consider but probably not always relevant.

[MDN documentation about setImmediate](https://developer.mozilla.org/en-US/docs/Web/API/Window/setImmediate).

brice-morin, 2018-11-01:
@fungiboletus if you have some snippets of generated code where is appears, feel free to mention it in this issue

fungiboletus, 2018-11-01:
Hei @brice-morin 
```
thing HelloThing {
	statechart HelloWorld init Greetings {
		state Greetings {
			transition -> Bye action print "Hello World!\n"
		}
		final state Bye {
			on entry print "Bye.\n"
		}
	}
}

configuration HelloCfg {
	instance my_instance: HelloThing
}
```

fungiboletus, 2018-11-01:
```javascript
HelloThing.prototype.build = function(session) {
	/*State machine (states and regions)*/
	/*Building root component*/
	this._statemachine = new StateJS.StateMachine('HelloWorld');
	let _initial_HelloThing_HelloWorld = new StateJS.PseudoState('_initial', this._statemachine, StateJS.PseudoStateKind.Initial);
	let HelloThing_HelloWorld_Greetings = new StateJS.State('Greetings', this._statemachine);
	let HelloThing_HelloWorld_Bye = new StateJS.FinalState('Bye', this._statemachine).entry(() => {
		console.log(''+'Bye.\n');
		setImmediate(()=>this._stop());
	});
	_initial_HelloThing_HelloWorld.to(HelloThing_HelloWorld_Greetings);
	HelloThing_HelloWorld_Greetings.to(HelloThing_HelloWorld_Bye).effect(() => {
		console.log(''+'Hello World!\n');
	});
}
```

brice-morin, 2018-11-01:
ah yes, in final states...

brice-morin, 2018-11-01:
So something like: `setTimeout(()=>this._stop(),0);` instead?

fungiboletus, 2018-11-01:
Yes, it works with this fix.

commit ab208313 (2018-11-01): fix #247
closed 2018-11-01 by brice-morin