TelluIoT/ThingML #259: How to run the tests ?

fungiboletus, 2018-11-13:
I'm working on #258 and I would like to run the tests for the NodeJS compiler on my machine.

However I have no idea about how to run the tests. I tried a `mvn clean install` in the `./testing` directory and it failed after 30s.

brice-morin, 2018-11-13:
OK, first pull my latest changes. I removed `Integer` some time ago because it is ambiguous  (compared to e.g. Int16), but it was quite intensively used in the tests...

you may have to `mvn clean install` in `./testing/framework` and `./testing/reportbuilder` first, then `mvn clean install` in `./testing`

brice-morin, 2018-11-13:
Also, it is possible to just run the nodejs tests, but I do not remember how...

brice-morin, 2018-11-13:
or in Eclipse, right click on testing project,-> run as JUnit

fungiboletus, 2018-11-14:
`cd ./testing/framework && mvn clean install`

```
 Failed to execute goal on project testing.framework: Could not resolve dependencies for project org.thingml:testing.framework:jar:2.0.0-SNAPSHOT: Failed to collect dependencies at org.thingml:compilers.framework:jar:2.0.0-SNAPSHOT -> org.thingml:thingml:jar:2.0.0-SNAPSHOT: Failed to read artifact descriptor for org.thingml:thingml:jar:2.0.0-SNAPSHOT: Could not find artifact org.thingml:language:pom:2.0.0-SNAPSHOT in thingml-snapshot (http://maven.thingml.org/thingml-snapshot/) -> [Help 1]
```

fungiboletus, 2018-11-14:
I did yesterday, but I just did it again to make sure and it doesn't help.

brice-morin, 2018-11-14:
Hmmm...

Try also `cd language && mvn clean install`

brice-morin, 2018-11-14:
well, the errors when compiling `language` are indeed not an issue (or at least not a blocking one).

As for the tests themselves, all the tests for which you have the proper setup to run the generated code (e.g. `node` and `npm` for NodeJS)  should run (the other should be skipped (probably posix, posixmt and go in your case?). Still, most nodejs tests should pass.... Not quite sure what is wrong, thoug... Is there some useful information from the errors and exceptions being displayed in you terminal?

fungiboletus, 2018-11-14:
Yes, posix and go are skipped. I probably don't have enough storage on my laptop to store the error log. I can't see anything in the terminal history, but this one is quite short. I might try again if I found it the alternative to `tee` in powershell and buy a 2TB SSD.

fungiboletus, 2018-11-14:
So the powershell equivalent to `tee` is `tee`. Example : `mvn clean install | tee -filepath bigdata.log`.

Using the latest version, by pulling and not fetching (fuck this fetch button on my Git GUI, so stupid), it runs. A few NodeJS tests fails, but that's expected I guess.

fungiboletus, 2018-11-14:
I also decided to ignore the many `Timeout while executing 'mvn exec:java'` because maven is so slow that it's not relevant.

closed 2018-11-14 by fungiboletus