TelluIoT/ThingML #263: Improve the documentation

fungiboletus, 2018-11-14:
- [ ] Up to date quick start
 - [ ] Single place to find helpful resources
 - [ ] More examples
 - [ ] Document annotations (see #215)
 - [ ] Explain more what ThingML is, can do, and what it can't do
 - [ ] How to contribute documentation
 - [ ] Syntax / keywords documentation
 - [ ] Make a website ?
 - [ ] Build an interactive tutorial that is actually cool, similar to [tryruby](https://ruby.github.io/TryRuby/) or equivalent.

brice-morin, 2018-11-14:
IMHO, we indeed need a website, acting as the single place to find helpful resources, documentation, getting started. I would suggest gh-pages. The README.md on the repo is just too messy, and I would only have info for developers on the master repo itself, not for users (who should use the website).

For the interactive tutorial, this is a good idea. A cheap version can be made quite easily (I guess), if we have the thingml compilers available somewhere in the cloud, + a way to run the generated code (packed in docker automatically generated from ThingML) on-demand. Well, actually not so cheap, maybe. A more fully-fledged version would need to use Language Server Protocol so as to get coloration, checks, etc. But I am not sure what support we get from XText there...

brice-morin, 2018-11-14:
Yes, I agree with you. Generated JS code executed in the local browser is the way to go.

At some point we had a prototopye of grammar for Atom (which had limitations). So no, not impossible to make something similar in a web-IDE. But we should see what we can get for free, first.

As for the visual state machine, that is also a good idea. We have discussed something similar in the ENACT project, but I am not allowed to work on this project :-) @nicolasferry can probably inform you on this matter.

fungiboletus, 2018-11-14:
If you find something that generates the grammar, sure. Personally I'm not planning to spend hours trying to find a tool to generate it, learn how to use, and debug it while it's so simple to do it by hand once in a while.

brice-morin, 2018-11-14:
The good thing if we manage to have LSP up and running, is that it is a rather standard way, and then we can get a VSCode plugin "for free", for example.

[See that for a XText/LSP/Monaco combo](https://github.com/svenefftinge/xtext-lsp-workshop)

fungiboletus, 2018-11-14:
Neat. Similar to ThingML in eclipse, it lacks variables names' completion in their running example but I guess it's something that could be fixed.

brice-morin, 2018-11-14:
Well, we can indeed expect the same limitations and the same benefits than the normal Eclipse plugin. With hopefully one single fix to implement to solve the problem in Eclipse and in the web editor. Of course, we need the LSP server running somewhere in the cloud, but we also need the ThingML compilers in the cloud... so not a big deal.

fungiboletus, 2018-11-14:
Oh so it needs to run some Java/XText things server side ?

brice-morin, 2018-11-14:
at least that is what I think, but I do not know much about LSP in practice...