TelluIoT/ThingML #264: [JS Compiler] Invalid generated code when a message parameter and the instances shares the same name

fungiboletus, 2018-11-16:
## ThingML model to reproduce the problem

```
thing fragment StatusMsg {
	message stalled(status: Boolean)
}
```

```
thing fragment StatusProvidedPort includes StatusMsg {
	provided port statusInput {
		receives stalled
	}
}
```

```
thing Status includes StatusProvidedPort {
    statechart init default {
        state default {
            internal event e.statusInput?stalled guard e.status == true action print "Stalled"
        }
    }
}
```

``` 
configuration Whatever {
    ...
    instance status:Status
    ...
    connector otherThing.statusInput => status.statusInput
    ...
}
```

## Generated code
	
```
controller.bus.on('statusInput?stalled', (status) => status.receivestalledOnstatusInput(status));
```

## Problem 

`status` is not the `status` instance but the `status` boolean parameter. In other words, a message parameter may override the instance reference.

brice-morin, 2018-11-16:
Are you fixing it?

fungiboletus, 2018-11-16:
It's very very... very far on my todo list. I just renamed the message
parameter.

commit 32d231fa (2018-12-11): fix #264
closed 2018-12-11 by brice-morin