TelluIoT/ThingML #265: The annotation checker does not seem to let anything compile

ffleurey, 2018-11-29:
I have nothing agains warnings at this point and errors later but I think that there is a bug and it now generated a lot of errors which should not be error.

Try compiling for example ThingML/doc/examples/02_ping.thingml

I get:

Compiling with "Plain Java"[Thu Nov 29 10:29:46 CET 2018] (Platform: java)
Selected input file: 02_ping.thingml (/home/franck/checkout/ThingML/doc/examples/02_ping.thingml)
Creating folder /home/franck/checkout/ThingML/doc/examples/thingml-gen/java/Ping
Checking configuration...
Checking configuration... Done! Took 4 ms.
Configuration Ping could not be compiled because of errors [Thu Nov 29 10:29:46 CET 2018].
	[line 12]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 13]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 14]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 16]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 20]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 21]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 22]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 24]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 28]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 29]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 30]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 32]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 36]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 37]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 38]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 40]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 44]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 45]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 46]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 48]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 52]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 53]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 54]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 56]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 60]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 61]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 62]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 64]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 68]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 69]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 70]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 72]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 76]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 77]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 78]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 80]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 84]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 85]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 86]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 88]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 92]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 93]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 94]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 96]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 100]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 101]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 102]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 104]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 108]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 109]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 110]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 112]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 116]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 117]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 118]: Annotation @js_type is invalid. @js_type applies to Type, . Specifies the concrete JavaScript type of a ThingML-defined type..
	[line 120]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 124]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 125]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
[... cut ...]
	[line 127]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..
	[line 130]: Annotation @java_type is invalid. @java_type applies to Type, . Specifies the concrete Java type of a ThingML-defined type..
	[line 132]: Annotation @c_type is invalid. @c_type applies to Type, . Specifies the concrete C type of a ThingML-defined type..
	[line 133]: Annotation @go_type is invalid. @go_type applies to Type, . Specifies the concrete Go type of a ThingML-defined type..

fungiboletus, 2018-11-29:
🍿

brice-morin, 2018-11-29:
Ah well... I announced it was a PoC :-)

Trying to fix now