TelluIoT/ThingML #267: The JavaScript compiler needs to be updated to support latest version of its state library

fungiboletus, 2018-12-03:
A major breaking change has been introduced in steelbreeze/state 7.2.0, see steelbreeze/state#12

fungiboletus, 2018-12-03:
I'm not sure but I feel like the internal events are broken.

brice-morin, 2018-12-03:
Anyway, since we do not have direct control over steelbreeze/state (though the guy is usually quite reactive at fixing issues), it is probably a good idea to be one minor revision behind in the generated JS code, unless we have a good reason to jump to the last version.

mesmo, 2018-12-04:
There was a big re-work of transitions - if you have some examples of the issues you're seeing I'll investigate and fix...

mesmo, 2018-12-05:
I have found a bug in the triggering of completion transitions after internal transition evaluation; v7.2.2 has been released and hopefully cures the issues you are seeing.

fungiboletus, 2018-12-05:
I confirm the bug has been fixed. We probably should update the generator to the new transition syntax at some point, but that's another issue to track.

Thanks again @mesmo

closed 2018-12-05 by fungiboletus

brice-morin, 2018-12-05:
Great! Once again, your reactivity on `state` is very much appreciated, @mesmo