TelluIoT/ThingML #268: Java generator ignores annotation on included Things

ffleurey, 2018-12-03:


brice-morin, 2018-12-03:
Indeed :-)

brice-morin, 2018-12-03:
looks at least like `@maven_dep` should not be problematic. Looking at the others now

brice-morin, 2018-12-03:
Actually, we should add this information in the annotation registry, whether or not a given annotation should be inherited (as I remember some discussions when in some cases, we did not want annotations to be inherited)

ffleurey, 2018-12-03:
I am not sure there are any left which we do not want to inherit. I think it was when we had fragments on the configurations that we had that pb... We should think about it.

ffleurey, 2018-12-05:
It does not seems to be solved.
Here is a test:

thing fragment ToInclude 
@maven_dep `		
	<dependency>
		<groupId>org.influxdb</groupId>
		<artifactId>influxdb-java</artifactId>
		<version>2.14</version>
	</dependency>`
	
@java_import `import org.influxdb.*;`
@java_import `import org.influxdb.dto.*;`
@java_import `import java.util.concurrent.TimeUnit;`
@java_import `import java.util.logging.Level;`
@java_import `import java.util.logging.Logger;`

@java_features `protected InfluxDB _influxDB = null;`

{
	
	function initdb() do
		println "Initilize the db"
	end
	
}

thing TestJavaInclude includes ToInclude 
{
	
	
	function doSomething() do
		if (`_influxDB == null`) initdb()
		
		`Point p = Point.measurement("gw_events")
            .time(System.currentTimeMillis(), TimeUnit.MILLISECONDS)
            .tag("host", "localhost")
            .addField("event", 1)
            .build();`
            
        println "do Something using the java library"
		
	end
	
	statechart init Init {
		
		on entry doSomething()
		
		state Init {}
	}
	
}

configuration TestJavaInclude {
	instance m : TestJavaInclude
}

brice-morin, 2018-12-05:
OK, seems like it works for `@java_import` and `@java_features`, but not for `@maven_dep`, unlike what I thought originally... Will fix that.

> Note that you can probably group your `@java_import` into one annotation, but anyway

commit 87f0c9ed (2018-12-03): fix #268 (to be tested, though :-)) by making @java_interface, @java_import and @java_features "inheritable"
commit c24e3533 (2018-12-05): fix #268 by also making @maven_dep inheritable
closed 2018-12-05 by brice-morin