TelluIoT/ThingML #271: 0ms timer from STL are ignored

fungiboletus, 2018-12-11:
`timer!timer_start(...,0)` are ignored because the standard implementation ignores 0ms or negative timeouts (`event m : timer?timer_start guard m.delay > 0`).
Should it be replaced by `>= 0` ?

ffleurey, 2018-12-11:
Should be fixed and deployed. Let me know if it works.

brice-morin, 2018-12-11:
Works with Node.JS (I haven't tested for the browser)

brice-morin, 2018-12-11:
Works with Java and Go, too

fungiboletus, 2018-12-11:
My guess : it's a conflict in the JS browser between the various implementations of setImmediate and the setTimeout of 0ms. I fixed by asking a timeout of 1ms minimum in before calling the timer.

ffleurey, 2018-12-11:
We could consider putting the 1ms hack in the library. i.e. replace the 0 with a 1. It is not acceptable to not get a timeout message after you have started a timer, but we never give guaranty exactly when you will get the message.

brice-morin, 2018-12-11:
Can't we just send the timeout message right away, without actually starting a JS timer with a 0ms period? Or do we need to rely on some side-effects of actually starting the JS timer?

fungiboletus, 2018-12-11:
I think in the ThingML context I would not expect to get the javascript side effect, just the timer to execute… so Yes right away is fine :-)

brice-morin, 2018-12-11:
Makes sense to me :-) I will test that out and commit if it indeed works