TelluIoT/ThingML #272: Formatting withe the ThingML Standalone pretty-printer

ffleurey, 2018-12-11:
org.thingml.xtext.helpers.ThingMLSerializer contains a pretty printer.

It works but the formatting is not pretty. It is using the default OneWhitespaceFormatter.
I do not understand why it does not use our ThingMLFormatter.

 Ctrl+Shift+F in eclipse and the saving of EMF Resources which works properly but I cannot find how to setup the ThingMLSerializer to use the ThingMLFormatter in the same way.

I give up for now!

ffleurey, 2018-12-11:
It could be that we should see if upgrading to the latest version of XText helps. I suspect some conflicts because the way XText handles formatting has changed at some point. We may still have some legacy code... not sure

TheDemonHuB, 2023-01-05:
is there any update?