TelluIoT/ThingML #277: Basic String operators are needed

brice-morin, 2019-01-25:
The `+` is sometimes used as a hack to concatenate strings in ThingML. This is not the intended usage of `+`, see #51. However, string concatenation is severely lacking in ThingML, though `+` is a useful hack as it works in at least JS and Java. 

Having a proper concatenation operator (not necessarily with the `+` syntax) would make it much easier to implement simple serializers in a PIM way in ThingML.

fungiboletus, 2019-04-05:
Here is some inspiration if you want to implement more operators than a fancy StringBuilder:

 * [Ruby](https://ruby-doc.org/core-2.6.2/String.html)
 * [Python](https://docs.python.org/3.7/library/stdtypes.html#string-methods)
 * [JavaScript](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/String#Methods_2)
 * [GoLang](https://golang.org/pkg/strings/)

Most of the methods could probably go into the STL. My personal preference is Ruby, but I would not mind doing it Go or JS (JS is a bit too Perl IMHO). Notice that I didn't link the Java doc'.

fungiboletus, 2019-04-05:
[More cool string methods from Ruby On Rails](https://api.rubyonrails.org/classes/String.html) for inspiration.

brice-morin, 2019-04-05:
The problem is the C generator, where it can be difficult to determine automatically when strings or dynamic structure should be freed... while other languages are garbage collected. And we probably do not want malloc and free in ThingML action language...