TelluIoT/ThingML #278: ThingML does not compile on MacOS or inside Docker

brice-morin, 2019-02-06:
@nicolasferry cannot compile ThingML on this Mac... :-)

To help him, since he's obviously not helped by his machine, I decided to make a Docker image, which basically clones the ThingML repo, compiles it, and somehow exposes the all-in-one jar. Well... it turns out I have the very same problem like he has (see below). Strangely, on a proper Linux machine, I am able to compile...

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$2 (file:/root/.m2/repository/com/google/inject/guice/3.0/guice-3.0.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$2
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
0    ERROR AbstractClassMirror - resource is empty: java:/Objects/java.lang.Object
java.lang.IllegalStateException
	at org.eclipse.xtext.common.types.access.impl.AbstractClassMirror.getEObject(AbstractClassMirror.java:94)
	at org.eclipse.xtext.common.types.access.TypeResource.getEObject(TypeResource.java:94)
	at org.eclipse.emf.ecore.resource.impl.ResourceSetImpl.getEObject(ResourceSetImpl.java:223)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:213)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:269)
	at org.eclipse.xtext.common.types.access.TypeResource.resolveJavaObjectURIProxy(TypeResource.java:176)
	at org.eclipse.xtext.common.types.impl.JvmTypeReferenceImplCustom.eResolveProxy(JvmTypeReferenceImplCustom.java:27)
	at org.eclipse.xtext.common.types.impl.JvmParameterizedTypeReferenceImpl.getType(JvmParameterizedTypeReferenceImpl.java:111)
	at org.eclipse.emf.mwe2.language.scoping.FactorySupport.findFactoriesCreationType(FactorySupport.java:46)
	at org.eclipse.emf.mwe2.language.scoping.FactorySupport.findFactoriesCreationType(FactorySupport.java:53)
	at org.eclipse.emf.mwe2.language.scoping.FactorySupport.findFactoriesCreationType(FactorySupport.java:37)
	at org.eclipse.emf.mwe2.language.scoping.Mwe2ScopeProvider.createComponentFeaturesScope(Mwe2ScopeProvider.java:128)
	at org.eclipse.emf.mwe2.language.scoping.Mwe2ScopeProvider.scope_Assignment_feature(Mwe2ScopeProvider.java:60)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.xtext.scoping.impl.AbstractDeclarativeScopeProvider.polymorphicFindScopeForReferenceName(AbstractDeclarativeScopeProvider.java:152)
	at org.eclipse.xtext.scoping.impl.AbstractDeclarativeScopeProvider.getScope(AbstractDeclarativeScopeProvider.java:100)
	at org.eclipse.xtext.linking.impl.DefaultLinkingService.getScope(DefaultLinkingService.java:58)
	at org.eclipse.xtext.linking.impl.DefaultLinkingService.getLinkedObjects(DefaultLinkingService.java:107)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:247)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:222)
	at org.eclipse.emf.ecore.resource.impl.ResourceSetImpl.getEObject(ResourceSetImpl.java:223)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:209)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:269)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eResolveProxy(BasicEObjectImpl.java:1477)
	at org.eclipse.emf.mwe2.language.mwe2.impl.AssignmentImpl.getFeature(AssignmentImpl.java:108)
	at org.eclipse.emf.mwe2.language.mwe2.impl.AssignmentImpl.eGet(AssignmentImpl.java:226)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eGet(BasicEObjectImpl.java:1011)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eGet(BasicEObjectImpl.java:1003)
	at org.eclipse.emf.ecore.util.EContentsEList$FeatureIteratorImpl.hasNext(EContentsEList.java:439)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolveCrossReferences(EcoreUtil.java:314)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolveAll(EcoreUtil.java:308)
	at org.eclipse.emf.mwe2.launch.runtime.Mwe2Runner.run(Mwe2Runner.java:86)
	at org.eclipse.emf.mwe2.launch.runtime.Mwe2Runner.run(Mwe2Runner.java:62)
	at org.eclipse.emf.mwe2.launch.runtime.Mwe2Runner.run(Mwe2Runner.java:52)
	at org.eclipse.emf.mwe2.launch.runtime.Mwe2Launcher.run(Mwe2Launcher.java:78)
	at org.eclipse.emf.mwe2.launch.runtime.Mwe2Launcher.main(Mwe2Launcher.java:36)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.codehaus.mojo.exec.ExecJavaMojo$1.run(ExecJavaMojo.java:282)
	at java.base/java.lang.Thread.run(Thread.java:834)
277  ERROR AbstractClassMirror - resource is empty: java:/Objects/java.lang.String
java.lang.IllegalStateException
	at org.eclipse.xtext.common.types.access.impl.AbstractClassMirror.getEObject(AbstractClassMirror.java:94)
	at org.eclipse.xtext.common.types.access.TypeResource.getEObject(TypeResource.java:94)
	at org.eclipse.emf.ecore.resource.impl.ResourceSetImpl.getEObject(ResourceSetImpl.java:223)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:213)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:269)
	at org.eclipse.xtext.common.types.access.TypeResource.resolveJavaObjectURIProxy(TypeResource.java:176)
	at org.eclipse.xtext.common.types.impl.JvmTypeReferenceImplCustom.eResolveProxy(JvmTypeReferenceImplCustom.java:27)
	at org.eclipse.xtext.common.types.impl.JvmParameterizedTypeReferenceImpl.getType(JvmParameterizedTypeReferenceImpl.java:111)
	at org.eclipse.xtext.common.types.util.JavaReflectAccess.getRawType(JavaReflectAccess.java:150)
	at org.eclipse.xtext.common.types.util.JavaReflectAccess.getParamTypes(JavaReflectAccess.java:144)
[... 90 lines cut ...]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.internalSwitch(Mwe2ExecutionEngine.java:66)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.internalApplyAssignments(Mwe2ExecutionEngine.java:142)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.inCase(Mwe2ExecutionEngine.java:114)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.internalSwitch(Mwe2ExecutionEngine.java:66)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.inCase(Mwe2ExecutionEngine.java:80)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.internalSwitch(Mwe2ExecutionEngine.java:66)
	at org.eclipse.emf.mwe2.language.factory.Mwe2ExecutionEngine.create(Mwe2ExecutionEngine.java:62)
	at org.eclipse.emf.mwe2.launch.runtime.Mwe2Runner.run(Mwe2Runner.java:93)
	... 10 more
```

brice-morin, 2019-02-06:
Seems like XText does not work with OpenJDK 11, which was the base image for the official maven image. Trying to downgrade to OpenJDK 10 (which does not seem to be a popular version... but it should be good enough for now for us)

brice-morin, 2019-02-06:
Failed also with OpenJDK 10 and 8. It seems there is another issue with XText: eclipse/xtext#1373

@ffleurey Seems like we need to upgrade XText :-)

ffleurey, 2019-02-06:
I am fine with it because now the new version is by default in the Eclipse distrib 😊

Franck

Provenance : Courrier pour Windows 10

De : Brice Morin
Envoyé le :mercredi 6 février 2019 15:46
À : TelluIoT/ThingML
Cc : Franck Fleurey; Mention
Objet :Re: [TelluIoT/ThingML] ThingML does not compile on MacOS or insideDocker (#278)

Failed also with OpenJDK 10 and 8. It seems there is another issue with XText: eclipse/xtext#1373
@ffleurey Seems like we need to upgrade XText :-)

brice-morin, 2019-02-06:
Well, I could give it a try this week, but I remember that task was not as simple as it should be (quite a few places where the version needs to be updated).... But I am very happy if anyone else volunteers :-)

brice-morin, 2019-02-07:
Well, that is a pain to migrate to Xtext 2.16 (the one integrated in Eclipse 2018-12). I now have to workaround that issue eclipse/xtext#1231

brice-morin, 2019-02-07:
Now fighting with eclipse/xtext-xtend#576

brice-morin, 2019-02-07:
Seems like the painful update to Xtext 2.16 helped. At least I was able to build and run ThingML within Docker.

closed 2019-02-07 by brice-morin

brice-morin, 2019-02-07:
Well, it seems there is still some problems compiling ThingML on MacOS... but the Docker images has successfully been tested on MacOS, providing an OK work-around :-)