TelluIoT/ThingML #280: UML visualization of Sessions

oysteinh, 2019-03-26:
I wanted to visualize my new example on sessions, but got the following runtime compiler error. I guess this is because you in this version have not yet visualized sessions? I did visualize sessions before...
 
 
Compiling with "Export to PlantUML"[Tue Mar 26 13:51:55 CET 2019] (Platform: uml)
Selected input file: config.thingml (C:\Users\oysteiha\Documents\Hiof\Undervisning\wrkspcThingML2019\HistoricSessions\config.thingml)
Creating folder C:\Users\oysteiha\Documents\Hiof\Undervisning\wrkspcThingML2019\HistoricSessions\thingml-gen\uml\HistoricPictures
Checking configuration...
Checking configuration... Done! Took 3 ms.
Infos(s) in C:\Users\oysteiha\Documents\Hiof\Undervisning\wrkspcThingML2019\HistoricSessions\config.thingml
       [line 6]: Dependency cycle: (usr, histsrv, db)
       [line 7]: Dependency cycle: (usr, histsrv, db)
       [line 8]: Dependency cycle: (usr, histsrv, db)
Please contact the ThingML development team (though GitHub's issue tracker) with 1) your input model, and 2) the following stack trace:
FATAL ERROR: Exeption calling ThingML Compiler: java.lang.UnsupportedOperationException: Region to be implemented
org.thingml.compilers.thing.common.FSMBasedThingImplCompiler.generateRegion(FSMBasedThingImplCompiler.java:67)
org.thingml.compilers.uml.PlantUMLThingImplCompiler.generateStateMachine(PlantUMLThingImplCompiler.java:143)
org.thingml.compilers.thing.common.FSMBasedThingImplCompiler.generateState(FSMBasedThingImplCompiler.java:39)
       org.thingml.compilers.uml.PlantUMLCompiler.compile(PlantUMLCompiler.java:87)
       org.thingml.compilers.uml.PlantUMLCompiler.do_call_compiler(PlantUMLCompiler.java:71)
       org.thingml.compilers.utils.OpaqueThingMLCompiler.compile(OpaqueThingMLCompiler.java:131)
       org.thingml.compilers.ThingMLCompiler.compile(ThingMLCompiler.java:267)
       org.thingml.eclipse.ui.commands.CompileThingFile.execute(CompileThingFile.java:255)
       org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:294)
       org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:96)
       sun.reflect.GeneratedMethodAccessor104.invoke(Unknown Source)
       sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
       java.lang.reflect.Method.invoke(Method.java:498)
       org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
       org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:320)
       org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:254)
    org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:164)
org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
       org.eclipse.core.commands.Command.executeWithChecks(Command.java:497)
  org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:490)
org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
org.eclipse.ui.internal.handlers.LegacyHandlerService.executeCommand(LegacyHandlerService.java:403)
org.eclipse.ui.menus.CommandContributionItem.handleWidgetSelection(CommandContributionItem.java:824)
       org.eclipse.ui.menus.CommandContributionItem.lambda$3(CommandContributionItem.java:801)
       org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
       org.eclipse.swt.widgets.Display.sendEvent(Display.java:4131)
       org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1055)
       org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3944)
       org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3547)
org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
       org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
  org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
       org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
       org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:636)
       org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
       org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
       org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
       org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:155)
       org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
  org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
       org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
       org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
       sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
       sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
       sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
       java.lang.reflect.Method.invoke(Method.java:498)
       org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
       org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
       org.eclipse.equinox.launcher.Main.run(Main.java:1501)

brice-morin, 2019-03-26:
Yes, it has been some changes in the inheritance relationships around Composite States, Regions and Sessions in the new metamodel. In the case of the UML compiler, if I remember well, Sessions were pretty much visualized like Regions (maybe with a note saying session or something). I should be able to fix that by Friday noon :-)

brice-morin, 2019-03-26:
OK, that was an easy one. Should be fixed and released to the update site.

commit 63fa2526 (2019-03-26): fix #280
closed 2019-03-26 by brice-morin