TelluIoT/ThingML #281: How to make a final composite state ?

fungiboletus, 2019-04-03:
```
composite state Lapin init Idle {
  final state Idle {}
}
```

It compiles with the following warning : `State Lapin is a sink state. Consider making it final or use @ignore "sink"`.

If I click on **Make State Final** it then changes it to `composite state final Lapin init Idle` which does not compile anymore.

ThingML version : not updated for months, perhaps it has been fixed recently.

ffleurey, 2019-04-03:
I think this it is the warning / suggested fix that is misleading. Composite states cannot be final. Brice what do you think?

brice-morin, 2019-04-03:
Well, the best would be to check the UML specs, but I also feel it is strange to have final composite states, since quite a lot can still happen with the composite state, even if there is no way to go out of it. I'll check if UML has an opinion on this, and fix this rule to only apply on atomic state (if they agree with us, or if they do not dare having an opinion)

brice-morin, 2019-04-03:
Well, anyway, there is no way we can make composite state final given the current metamodel. So I will update this validation rule so that we are not suggesting impossible things :-)

commit bbd5bcb5 (2019-04-03): fix #281 (check and re-open if not :-))
closed 2019-04-03 by brice-morin