TelluIoT/ThingML #283: The state machine dependency has breaking changes

fungiboletus, 2019-04-04:
As stated here in https://github.com/steelbreeze/state/blob/master/RELEASES.md#v800 steelbreeze/state has breaking changes in its 8.0.0 version.

Someone confident with ThingML should probably run the unit tests with this version.

brice-morin, 2019-04-04:
Do you mean the `package.json` we generate uses by default this new 8.0.0 version? If so, I would first correct that, so as it use a version we know works (in principle, the latest 7.x.y).

Then, we can talk about migrating to 8.x.y :-)

fungiboletus, 2019-04-04:
No it does not. We should discuss about the migration.

brice-morin, 2019-04-04:
That was what I feared :-)

brice-morin, 2019-04-10:
OK, work in progress :-)

Quite a big refactoring in the end, but so far I like the new generated code (though it does not work at this very moment, but I am getting closer)

commit 7db8763a (2019-04-11): fix #283 + some other improvements to JS compilers
closed 2019-04-11 by brice-morin