TelluIoT/ThingML #284: Signature in Messages?

zeinebBABACHEIKH, 2019-04-29:
Hi!
Is it possible to transfer a variable between things via messages?
In my case, I need to send the temperature value from the sensor thing to the actuator thing.
If not, is there any other way to do so?
Thanks

brice-morin, 2019-04-29:
If this variable is a simple `datatype` (int, float, etc), it should be no problem to send it in a message. If the variable is a more complex `object`, this is less certain: it will probably work in Java and JavaScript (and maybe Go), but it will not work in C.

My best advice is try and then comeback to us if it does not work or if you need more help :-)

zeinebBABACHEIKH, 2019-04-29:
I tried it. the problem is still there. 
How can the message contains a variable? should I change it into a function to return a variable? if so, I have to define the function in both things? Now I read the variable in a thing (sensor) but I will need that variable in another thing (actuator)

brice-morin, 2019-04-30:
Well, it does not directly contain a variable. See code snippet below:

```
thing XXX {
    message myMessage(a : UInt8) //define a message with a parameter that has the same type as your variable

    required port myPort {
        sends myMessage
    }

    property myVar : UInt8  //the variable that contains the temperature value (could also be a local variable). Adjust type as needed

    ...
    myPort!myMessage(myVar) //sends the message with the current value of myVar, using a reference to myVar
    ...
}
```

zeinebBABACHEIKH, 2019-05-01:
This is what I exactly did. Now my problem is the receiver, How can the receiver read the value from the message?

brice-morin, 2019-05-01:
you need to do something like that:

```
thing Receiver {
  property receiverVar : UInt8
...
  statechart init INIT {
    state INIT {
        internal event e : myPort?myMessage
        action do 
            print e.myVar
            receiverVar = e.myVar
    }
  }
}
```

closed 2019-05-01 by brice-morin