TelluIoT/ThingML #286: Implement a Rust compiler

fungiboletus, 2019-05-08:
@vetlevo and I are trying to convince @ffleurey to use a modern programming language to replace his C deployments whenever possible. Safety and more specifically memory safety are the main concerns with C and C++.

We are facing some resistance with moving towards Go, Java, or JavaScript. Java or JavaScript are too heavy and not low level enough. Go is perhaps too simple and still has a garbage collector.

At the same time,

> Rust was the "most loved programming language" in the Stack Overflow Developer Survey for 2016, 2017, 2018, and 2019.

It [offers similar performances to C and C++](https://benchmarksgame-team.pages.debian.net/benchmarksgame/faster/rust.html) while offering more memory safety (as long as you don't use the `unsafe` keyword) and a more modern API.

We think it could be a good fit for ThingML and the Rust compiler could replace the C compiler with only positive impacts in many deployments.

brice-morin, 2019-05-08:
+ it has a concurrency model aligned with ThingML with message passing between threads (a bit like Go), which would remove the need for generating the internal serialization/deserialization whenever we send messages in ThingML (even between local instances). Less code to generate, less code to maintain in the compilers. 

+ the C family of compilers starts to be "crowded". It could be the occasion to only focus the C compiler on micro-controllers only, and clean a lot of things there....

So, I like the idea :-)

One question is whether we should go for a full generative approach like in C compiler, or a "semi-generative" approach like in other compilers, which rely on a framework for state machines. 

Also, the biggest question is who is going to implement it :-)

fungiboletus, 2019-05-13:
> Also, the biggest question is who is going to implement it :-)

It's a very good question. At Tellu we do not know Rust yet.

brice-morin, 2019-05-13:
Let's wait for a concrete use case, and a common project for this use case :-)

Happy @ffleurey ? But we keep the issue open ;-)

fungiboletus, 2019-05-20:
I did my first Rust Hello World yesterday, so things are going pretty well I would say.

brice-morin, 2019-05-20:
Good, good :-) We might discuss that around lunch tomorrow as I'll be at Tellu in the morning for another project.