TelluIoT/ThingML #290: How to register annotations ?

fungiboletus, 2019-10-18:
```Annotation @nodejs_package is not registered. Non-registered
   annotations will be deprecated in a future version.```

brice-morin, 2019-10-18:
You can follow the same procedures as for the others: https://github.com/TelluIoT/ThingML/blob/master/language/thingml/src/org/thingml/annotations/AnnotationRegistry.java

fungiboletus, 2019-10-18:
I tried without success:

```java
annotations.put("nodejs_package", new Annotation(
		"nodejs_package",
		"If present, the configuration is compiled as a NodeJS NPM package",
		new EClass[] {ThingMLPackage.eINSTANCE.getConfiguration()}
));
```

and

```
configuration  Canard
@nodejs_package
{
...
```

Annotation is still marked as unregistered :(

brice-morin, 2019-10-18:
Hmmm.... strange. Try compiling from root and from ./language

fungiboletus, 2019-10-18:
I did. Didn't really mind. However I since rebooted for another reason and I can see that it's working.

closed 2019-10-18 by brice-morin