TelluIoT/ThingML #291: Why is SourceBuilder deprecated ? What should it be instead ?

fungiboletus, 2019-10-18:


brice-morin, 2019-10-18:
It is an extremly neat utility, basically building an in-memory model of the generated code, and then relying on this in-memory model to manage indentation, etc.

The main issue is that it introduces an indirection between the ThingML model and the generated code, which IMHO makes compilers harder to maintain.

We do not have plan to refactor the existing compilers using SourceBuilder. If you modify an existing compiler using SourceBuilder, well, I guess you should continue using it.

So, unless you plan to implement a compiler for Python (where proper indentation is necessary in the generated code), I would advise against using SourceBuilder for new compilers. Instead look at Java or C compilers (basically just relying on StringBuilder).

fungiboletus, 2019-10-28:
I think it's nice to have good indentation. We could perhaps run `eslint --fix` as a step if we decide to remove indentation from the generated code in Java.

brice-morin, 2019-10-28:
Yes, it is nice to have good indentation. But as you point out, there are ways of automatically indent the code as a post-processing step, through tools and in the IDEs. That is the way we should investigate, so that we can keep the compilers maintainable, with a straightforward mapping between the compilers and the generated code.

I guess for Java, we could look into https://github.com/revelc/formatter-maven-plugin 

For C, there is also a bunch of tools.

It is probably good to have those indentation tools inside the `package.json`, `POM.xml`  or similar, so that they can be installed when "compiling". We should probably not assume these tools will be available beforehand on the machine. If we assume that, we should make sure indentation is an optional step that will not crash if the tool is not available,