TelluIoT/ThingML #297: Updating the Xtext Community Website

miklossy, 2020-02-04:
Hello ThingML team,

The Xtext team would like to update the [Xtext community website](https://www.eclipse.org/Xtext/community.html) listing all the cool projects people are building around Xtext/Xtend.

See also the corresponding [GitHub issue](https://github.com/eclipse/xtext/issues/1541) and [Bugzilla ticket](https://bugs.eclipse.org/bugs/show_bug.cgi?id=328477).

If you are interested that the ThingML  project is listed there, please provide the following information as a comment to this issue:

```
<tr>
	<td><a href="project_link">project_name</a></td>
	<td>project_description_max_2_sentences</td>
	<td>license</td>
	<td>category</td>
	<td>author(s)</td>
</tr>

```
We will then update the Xtext community website and get it online as soon as possible.

Thanks!
Tamás

brice-morin, 2020-02-04:
Hi,

Thank you for your interest in ThingML. It seems like a very good idea to list ThingML on the XText community website. 

What do you think @ffleurey ?

Cheers,

Brice.

miklossy, 2020-12-09:
Hello @brice-morin / @ffleurey !

Any update on this?

fungiboletus, 2021-04-09:
@miklossy Thanks for insisting. I suggest the following description : let me know if you want any change or if you want me to do a pull request on your project.

```html
<tr>
	<td><a href="https://github.com/TelluIoT/ThingML">ThingML</a></td>
	<td>ThingML is a modelling language, a set of tools, and a methodology. It includes editors and an advanced multi-platform code generation framework, which support multiple target programming languages (C, Java, JavaScript, Go).</td>
	<td>Apache 2 License</td>
	<td>Language, Framework</td>
	<td>Franck Fleurey, Brice Morin</td>
</tr>
```

miklossy, 2021-04-16:
The `ThingML` project has been added to the [Xtext community website](https://www.eclipse.org/Xtext/community.html).

closed 2021-04-16 by miklossy