TelluIoT/ThingML #301: http://thingml.org/dist/update2 is not reachable

easyvolts, 2020-08-29:
I just tried to access the web page with eclipce pluging http://thingml.org/dist/update2 and it doesn't work.

brice-morin, 2020-08-29:
This server is down and won't be up any time soon (never). You can find the update sites here: https://github.com/TelluIoT/ThingML/releases (thingml.repository*.zip), which you can download and install in Eclipse using local update site.

closed 2020-08-29 by brice-morin

easyvolts, 2020-08-30:
Brice-morin, thank you for the fast response. I'd like to let you know that documentation in this chapter https://github.com/TelluIoT/ThingML#thingml-eclipse-based-ide needs to be fixed to reflect the information from above.
Best regards, Valerii

brice-morin, 2020-08-30:
Good catch, thanks! It should be fixed now.